koide3/hdl_graph_slam #120: Errors compiling ndt_omp

amlobat2, 2020-04-26:
I am new to ROS and Linux and can't seem to figure out why I am getting issues when using catkin_make to compile the ndt_omp requirement. The error I get is below. 

`In file included from /media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/src/pclomp/gicp_omp.cpp:2:0:
/media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp: In member function ‘void pclomp::GeneralizedIterativeClosestPoint<PointSource, PointTarget>::computeCovariances(typename pcl::PointCloud<PointTDiff>::ConstPtr, typename pcl::search::KdTree<PointT>::ConstPtr, pclomp::GeneralizedIterativeClosestPoint<PointSource, PointTarget>::MatricesVector&)’:
/media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp:64:51: error: there are no arguments to ‘omp_get_max_threads’ that depend on a template parameter, so a declaration of ‘omp_get_max_threads’ must be available [-fpermissive]
   std::vector<std::vector<int>> nn_indecies_array(omp_get_max_threads());
                                                                                 ^~~~~~~~~~~~~~~~~~~
/media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp:64:51: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp:65:52: error: there are no arguments to ‘omp_get_max_threads’ that depend on a template parameter, so a declaration of ‘omp_get_max_threads’ must be available [-fpermissive]
   std::vector<std::vector<float>> nn_dist_sq_array(omp_get_max_threads());
                                                                                  ^~~~~~~~~~~~~~~~~~~
media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp:69:43: error: there are no arguments to ‘omp_get_thread_num’ that depend on a template parameter, so a declaration of ‘omp_get_thread_num’ must be available [-fpermissive]
     auto& nn_indecies = nn_indecies_array[omp_get_thread_num()];
                                                                      ^~~~~~~~~~~~~~~~~~
/media/oliver/amlobat2_files/hdl_graph_slam/src/ndt_omp/include/pclomp/gicp_omp_impl.hpp:70:41: error: there are no arguments to ‘omp_get_thread_num’ that depend on a template parameter, so a declaration of ‘omp_get_thread_num’ must be available [-fpermissive]
     auto& nn_dist_sq = nn_dist_sq_array[omp_get_thread_num()];
                                                                  ^~~~~~~~~~~~~~~~~~

`

I get this same error for each instance that omp_get_max_threads and omp_get_thread_num. Can someone please help me figure out what to do? Thanks!!

fghjkl17, 2020-05-02:
I want to know which architecture does your CPU belong to? amd64 or arm64 or else?

amlobat2, 2020-05-03:
I'm working on an NVIDIA Xavier so it's armv8

fghjkl17, 2020-05-04:
Could you change line 5-6 in [hdl_graph_slam/CMakeLists.txt](https://github.com/koide3/hdl_graph_slam/blob/master/CMakeLists.txt) and [ndt_omp/CMakeLists.txt](https://github.com/koide3/ndt_omp/blob/master/CMakeLists.txt)
from
`add_definitions(-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2)`
`set(CMAKE_CXX_FLAGS "-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2")`
to
`add_definitions(-std=c++11)`
`set(CMAKE_CXX_FLAGS "-std=c++11")`
and then delete build and devel folders and catkin_make again.

ipinyulun, 2020-05-26:
[error_Tx2.txt](https://github.com/koide3/hdl_graph_slam/files/4680820/error_Tx2.txt)

> Could you change line 5-6 in [hdl_graph_slam/CMakeLists.txt](https://github.com/koide3/hdl_graph_slam/blob/master/CMakeLists.txt) and [...]

i encunter the same issue (as file uploaded up there)
and i try this solution, however it does not work.
one of my concern is  PCL1.8.
becoz im running on ubuntu 18.04 on Jetson Tx2.
aarch64

so i was unable to install PCL 1.7

therefore, is the cause of the error becoz of PCL1.8 ? 
if yes, how to install PCL 1.7 into Jetson Tx2?
google doesnt seem help as well

ipinyulun, 2020-05-26:
> [error_Tx2.txt](https://github.com/koide3/hdl_graph_slam/files/4680820/error_Tx2.txt)
>
>
> i encunter the same issue (as file uploaded up there)
> and i try this solution, however it does not work.
> one [...]

issue solved at https://github.com/koide3/ndt_omp/issues/2

FishInWave, 2020-08-26:
@ipinyulun 
Hi, thanks for your quote. 
I successfully run hdl_graph_slam on my Jetson Xavier with a rosbag recorded by a robosense-16 LiDar. But I note that the SLAM cannot run in 1X speed normally even when Xavier is in "MODE 30W ALL"，i.e. 8-core is activeted, and I must use `record paly -r 0.5` to slow down play speed.
I wonder whether you can run this program on your TX2 in real time?