koide3/hdl_graph_slam #121: GPS time instead of ros time

Johntec, 2020-05-05:
Hi @koide3 , is it possible to have and set a parameter to use the GPS time for all messages published instead of the ROS time when the GPS time is available?

how in General the time synchronization between different sensors (gps, laser) and system is calculated.

Thanks for the good work.

koide3, 2020-05-07:
Hi @Johntec ,
We don't take care of synchronization because it is rather a hardware mechanism. We just assume the message timestamps are already synchronized, and associate messages with close timestamps. If you know the GPS time of each message with some hardware PPS interface, you can just replace ```msg.header.stamp``` with it.

Johntec, 2020-06-03:
I have done that and it worked, thank you.

closed 2020-06-03 by Johntec