koide3/hdl_graph_slam #122: Pose graph format generated from service dump

nyangshawbin, 2020-05-13:
Dear @koide3,

I would like to clarify if the pose graph generated (graph.g2o) from the rosservice /hdl_graph_slam/dump is the optimized pose graph after loop closure. 

If so, may i know what format is each line of the output in? Following is a snippet:
`VERTEX_SE3:QUAT 3 -0.421949 0.820959 0.00217435 -0.0177581 -0.00476316 0.906195 0.422459 
VERTEX_SE3:QUAT 4 -1.24384 1.42655 0.0073593 -0.015512 0.00142036 0.988524 0.150256 
VERTEX_SE3:QUAT 5 -2.27197 1.49608 0.00940595 -0.0131559 0.00596053 0.999626 0.0232131 ` 

I would like to compare this optimized trajectory with another trajectory in TUM format: 
`timestamp tx ty tz qx qy qz qw` using the [EVO](https://github.com/MichaelGrupp/evo) repo.

Any help would be appreciated. Thank you in advance.

koide3, 2020-05-14:
Hi @nyangshawbin ,

One line about an SE3 vertex indicates ```vertex_type vertex_id x y z qx qy qz qw```. You can see [this page](https://github.com/RainerKuemmerle/g2o/wiki/File-Format) and [this](https://github.com/RainerKuemmerle/g2o/wiki/File-format-slam-3d) for the detail of g2o's graph format.

nyangshawbin, 2020-05-14:
Appreciate for the help, thanks!

closed 2020-06-01 by nyangshawbin

wakasu, 2021-08-31:
@nyangshawbin 
Hey, what approach you used to accomplish this? how you implemented the timestamp data?
I'm also trying to do comparison between multiple estimations and ground truth using EVO repo.

Really appriciate any help 
Regards,

PrePLEX23, 2023-01-09:
------------hdl_graph_slam_nodelet.cpp----------------
bool save_traj_service(hdl_graph_slam::SaveTrajRequest& req, hdl_graph_slam::SaveTrajResponse& res) {
std::vector<KeyFrameSnapshot::Ptr> snapshot;

keyframes_snapshot_mutex.lock();
snapshot = keyframes_snapshot;
keyframes_snapshot_mutex.unlock();

outFile2.open(req.destination, std::ios::out | std::ios::trunc);
for(KeyFrameSnapshot::Ptr kf : snapshot){
// cloud->header.stamp = snapshot.back()->cloud->header.stamp;
double pose_stamp = kf->cloud->header.stamp/1000000;
Eigen::Matrix4f apose = kf->pose.matrix().cast<float>();
float tx = apose(0,3);
float ty = apose(1,3);
float tz = apose(2,3);
Eigen::Quaternionf pose_quat(apose.block<3,3>(0,0));
float qw = pose_quat.w();
float qx = pose_quat.x();
float qy = pose_quat.y();
float qz = pose_quat.z();
outFile2<<pose_stamp<<","
<<tx<<","<<ty<<","<<tz<<","
<<qx<<","<<qy<<","<<qz<<","<<qw;
outFile2<<std::endl;
}
res.success = true;

return true;
}
------------------------------------------------
add above code to hdl_graph_slam_nodelet.cpp, check and modify the code related to srv.Call this service, you can get a trajectory for evo repo.

jonasky, 2023-08-28:
> 
Is it possible to get more detailed instructions on this?
i tried implementing the code but i recieved erros messages regarding it. my c++ skills are not up to date unfortunately