koide3/hdl_graph_slam #123: Occupy too many computer resources

FishInWave, 2020-05-15:
hi, @koide3 
When I run the demo in README.MD as follow:

`roslaunch hdl_graph_slam hdl_graph_slam_400.launch
rosparam set use_sim_time true
roscd hdl_graph_slam/rviz
rviz -d hdl_graph_slam.rviz
rosbag paly --clock hdl_400.bag
`
It works, and show the map.
It's normal while rosbag is playing。
But  a few minutes later after rosbag play finished, the computer become too laggy, even though I didn't do anything.
Finally, I have to restart my PC.
My PC's details show as follow

> Memory 15.5GiB
>Processor Intel Core i7-7820HK CPU @2.90GHz*8
>Graphics : Intel HD Graphics 630(Kaby Lake GT2)

I guess that there are some operations running all the time, even if there are no new data. And these operations comsume the memory or processor resource of PC. But I haven't found the source of the problem .
Hoping for your help, thanks。

FishInWave, 2020-05-15:
> hi, @koide3
> When I run the demo in README.MD as follow:
>
> `roslaunch hdl_graph_slam hdl_graph_slam_400.launch`
> `rosparam set use_sim_time true`
> `roscd hdl_graph_slam/rviz`
> `rviz -d [...]

I have found the root of this problem.
/map_points topic is published per 10 seconds, this is useful while running.
However, when data flows have already finished, function `map_points_publish_timer_callback` is still  called over and over, and then I notice the occupied memory is increasing at the same time.
when I dis-subscribe this topic in rviz, the memory is freed, and computers run nomally and smoothly.
The possible solution is that let `map_points_publish_timer_callback` only work while data flow still run.

I haven't try this online yet, just test on rosbag. 

By the way, I focus on this problem because I want to run hdl_graph_slam all the time to update map realtimely, and then location robots(itself) and route planning.

koide3, 2020-05-18:
Hi @FishInWave ,
Thanks for reporting the issue. As you pointed out, the current implementation publishes the map cloud every 10 seconds even if the data flow has been stopped, and it leads to consuming the resource.
I believe it's not difficult to fix this problem by checking if the system received a new data and advertising the map topic as a latched topic. I'll do it in a few days.

koide3, 2020-05-19:
@FishInWave 
I uploaded the updated code. Can you pull the devel branch and check if the problem is fixed?

FishInWave, 2020-05-19:
Hi @koide3 
I have tested it on hdl_400.bag, and the problem has been fixed !

PR #124 (2020-05-19): Fix resource consuming problem 
- Fix #123 
- IMU-based frontend (Preliminary)
closed 2020-05-19 by FishInWave