koide3/hdl_graph_slam #125: Question about the use of quaternion in IMU deskewed

FishInWave, 2020-05-20:
hi @koide3 
I have a question about the use of  quaternion in prefiltering_nodelet.cpp.The  code involved shows as below:
`      double delta_t = scan_period * static_cast<double>(i) / cloud->size();
      Eigen::Quaternionf delta_q(1, delta_t / 2.0 * ang_v[0], delta_t / 2.0 * ang_v[1], delta_t / 2.0 * ang_v[2]);
      Eigen::Vector3f pt_ = delta_q.inverse() * pt.getVector3fMap();`
The usual method I have learned to use a quatermion is  "q=[cos(theta/2), nx*sin(theta/2), ny*sin(theta/2), nz*sin(theta/2)]", where nx,ny,nz is the vector of rotation axis, and theta represents rotation angle.
So can you please explain above code? 
Thank you.

koide3, 2020-05-21:
Hi @FishInWave ,
It's a well-known approximation technique to compute a delta quaternion from angular velocity. See [this](https://stackoverflow.com/a/24201879/3950982).

FishInWave, 2020-05-26:
@koide3 
 Thank you for your reply !!!  : )

closed 2020-05-26 by FishInWave