koide3/hdl_graph_slam #135: Error when calling rosservice

zxl19, 2020-07-13:
Hi, thank you for your excellent work!
After running hdl_graph_slam, I tried the following command to save data for interactive_slam:
```shell
rosservice call /hdl_graph_slam/dump
```
However, I got the following error message:
```shell
ERROR: Unable to load type [hdl_graph_slam/DumpGraph].
Have you typed 'make' in [hdl_graph_slam]?
```
I tried calling the rosservice to save map, and got this error message as well. What should I do specifically to solve this?

Best regards.

zxl19, 2020-07-13:
Also, I used your forked version of LeGO-LOAM-BOR. I'm using a Robosense-32 LiDAR, the vertical angular resolution is not fixed(the lines are not evenly distributed vertically), I wrote an implementation to project the points correctly in `imageProjection.cpp` and modified the config file accordingly. Still, the result is poor. A lot of points are missing and the map cannot be built successfully. But this implementation works well in the original LeGO-LOAM, I still can't find what seems to be the problem, can you give me a hint?

koide3, 2020-07-14:
Probably the first issue can be solved by https://github.com/udacity/simple_arm/issues/6

Regarding LeGO-LOAM-BOR, since I'm not the original author of that package, I cannot provide useful advices. Please take a look at the original repo: https://github.com/facontidavide/LeGO-LOAM-BOR.

zxl19, 2020-07-14:
Thank you for your reply!

> Probably the first issue can be solved by udacity/simple_arm#6

I tired the following command:
```shell
 rosservice call /hdl_graph_slam/save_map "resolution:0.05 destination:/home/test.pcd'"
```

and got the following error:

```shell
ERROR: Incompatible arguments to call service:
Not enough arguments:
 * Given: ["resolution:0.05 destination:'/home/zhangxl/test.pcd'"]
 * Expected: ['utm', 'resolution', 'destination']
Provided arguments are:
 * resolution:0.05 destination:'/home/zhangxl/test.pcd' (type str)

Service arguments are: [utm resolution destination]
```

I used `Tab` to autocomplete and got:

```shell
rosservice call /hdl_graph_slam/save_map "utm: false
resolution: 0.0
destination: ''" 
```

I completed the command and it's working. Perhaps you could update this part in the README file because it's a little bit different.

> Regarding LeGO-LOAM-BOR, since I'm not the original author of that package, I cannot provide useful advices. Please take a look at the original repo: https://github.com/facontidavide/LeGO-LOAM-BOR.

I will turn to him for help, thanks.

Best wishes.

closed 2020-07-14 by zxl19