koide3/hdl_graph_slam #136: Improving loop closure for large drift

nyangshawbin, 2020-07-13:
Hi, 

I would like to get some advice on tuning the parameters especially for loop closure for large drifts. I am running SLAM offline on a rosbag recorded from Gazebo. Loop closure seems to be working well for me for small drifts as shown in the first picture below. However, for larger drift, as shown in second pic, the loop closure was not good. 

[image]
[image]

I have tried increasing the `<param name="distance_thresh" value="15.0" />` to increase the loop closure search window to correct larger drifts but the result was instead sub-optimal as improper loop closure was done. Hence would like to ask if there are other parameters which i can tune to improve the loop closure performance, or if im doing something wrong here. Thank you in advance! Following are more information about my setup/launch file.

Setup:
Lidar: HDL-32E
[hdl_graph_slam_400 launch file](https://github.com/nyangshawbin/outdoor_slam/blob/master/hdl_graph_slam/launch/hdl_graph_slam_400.launch)

koide3, 2020-07-14:
The loop closing implementation is very basic and limited, and suffer from large drift like the one in your case. If you're going to do outdoor mapping, I recommend using a GPS that greatly helps to close large loops at outdoor.

nyangshawbin, 2020-07-14:
Hi, thanks for getting back. I see, i will consider that advice then thanks once again!

closed 2020-07-14 by nyangshawbin