koide3/hdl_graph_slam #137: Using IMU

nyangshawbin, 2020-07-19:
Hi @koide3 , thanks for the work. I would like to utilize HDL graph slam to 3d map an outdoor environment as i noticed the result in z-axis is much easier to tune as compared to other 3D SLAM techniques. 

However, I would like to improve the local SLAM, especially to minimize the drift as much as possible so loop closure will be easier. You have suggested to use gps on the previous [issue](https://github.com/koide3/hdl_graph_slam/issues/136#issuecomment-657901248) i enquired, I have tried it in a simulated environment and the result was really good.

But i would like to leverage more on IMU, hoping to minimize the drift. I see that theres an [imu subscription](https://github.com/koide3/hdl_graph_slam/blob/701a39e17e2be6d39261b5f27e0d1df993efbc99/apps/hdl_graph_slam_nodelet.cpp#L111) in the hdl_graph_slam nodelet, do i simply remapped the `/gpsimu_driver/imu_data` to my own `/imu`. If so, is IMU used in the odometry estimation as well or isit only used in the graph slam node?

nyangshawbin, 2020-07-19:
Just to add, i have remapped the `/imu` topics and the map got better. However, at certain areas of the gazebo simulation there are less features and hence lidar-based odometry is not as accurate. I have tried tweaking the `imu_orientation_edge_stddev` and `imu_acceleration_edge_stddev` hoping to give more weightage to the IMU , but the robot still seems to be 'slipping' in such conditions.

koide3, 2020-07-21:
Hi @nyangshawbin ,
In this package, IMU information is used only in the SLAM backend. The frontend can take advantage of IMU if you set ```enable_imu_frontend``` to true, but it is done in a so-called loose coupling manner with [ethzasl_sensor_fuction](http://wiki.ros.org/ethzasl_sensor_fusion), and I suppose it doesn't help to correct the slipping effect. Anyways, if you want to try, you can take a look at [hdl_graph_slam_imu.launch](https://github.com/koide3/hdl_graph_slam/blob/master/launch/hdl_graph_slam_imu.launch).

nyangshawbin, 2020-07-21:
Hi @koide3 , thank you for getting back. I would like to clarify, if IMU can be used through the ethzasl_sensor_fusion package for better odometry estimation, wouldn't it help in the slipping effect? 
Nevertheless, thanks for the reference, i will try out the parameter to see if it helps!

koide3, 2020-07-24:
It would somehow reduce the slipping effect with better IMU-based initial guess for scan matching. But I suppose the improvement would not be significant because the loose coupled LIDAR-IMU fusion inherently assumes decent scan matching results and cannot deal with scan matching corruption.

nyangshawbin, 2020-07-27:
I see. Thank you so much,  I shall close the issue.

closed 2020-07-27 by nyangshawbin