koide3/hdl_graph_slam #138: ERROR: cannot launch node of type [msf_updates/pose_sensor]: msf_updates

CAKGOD, 2020-08-06:
system: ubuntu 18.04 
ros: melodic
problem: 
ERROR: cannot launch node of type [msf_updates/pose_sensor]: msf_updates
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/cakgod/hdl_graph_slam/src
ROS path [2]=/home/cakgod/ndt_omp/src
ROS path [3]=/opt/ros/melodic/share

 Is there anyone who can help solve this problem? Thank you~~

FishInWave, 2020-08-06:
I met this issue just half an hour ago in ros-melodic，and it just disappeared after `reboot` .

CAKGOD, 2020-08-07:
@FishInWave   This problem still exist after reboot for me. But it causes nothing in my work so i close it. ~~~~

closed 2020-08-07 by CAKGOD

don-thomas, 2021-10-07:
I was able to fix this problem by doing the following:

```
sudo apt install libgoogle-glog-dev 
cd ~/ros_workplace/src
git clone https://github.com/ethz-asl/ethzasl_msf.git
cd .. && catkin_make -DCMAKE_BUILD_TYPE=Release
```