koide3/hdl_graph_slam #139: How to run hdl_graph_slam on Ubuntu18.04 with pcl-1.8

FishInWave, 2020-08-06:
My enviorment is Ubuntu18.04, ros-melodic and pcl-1.8.
When I roslaunch hdl_graph_slam_501.launch，it failed to load nodelet. According to the ERROR message, ** Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)** may means this error caused by pcl version.
When I copy the code to another laptop which has the same enviorment(Ubuntu18.04, ros-melodic and pcl-1.8.),and compile, `roslaunch` run without any error.
I have no idea where is the root of this proble. 

> [ERROR] [1596707127.917670153]: Failed to load nodelet [/hdl_graph_slam_nodelet] of type [hdl_graph_slam/HdlGraphSlamNodelet] even after refreshing the cache: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libhdl_graph_slam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [ERROR] [1596707127.917705797]: The error before refreshing the cache was: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libhdl_graph_slam_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [FATAL] [1596707127.917859921]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
> [ERROR] [1596707127.945973122]: Failed to load nodelet [/scan_matching_odometry_nodelet] of type [hdl_graph_slam/ScanMatchingOdometryNodelet] even after refreshing the cache: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libscan_matching_odometry_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [ERROR] [1596707127.946050005]: The error before refreshing the cache was: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libscan_matching_odometry_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [FATAL] [1596707127.946293424]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
> [ERROR] [1596707127.972745446]: Failed to load nodelet [/floor_detection_nodelet] of type [hdl_graph_slam/FloorDetectionNodelet] even after refreshing the cache: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libfloor_detection_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [ERROR] [1596707127.972814259]: The error before refreshing the cache was: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libfloor_detection_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [FATAL] [1596707127.973003433]: Failed to load nodelet '/floor_detection_nodelet` of type `hdl_graph_slam/FloorDetectionNodelet` to manager `velodyne_nodelet_manager'
> [ERROR] [1596707128.000253043]: Failed to load nodelet [/refine_map_nodelet] of type [hdl_graph_slam/RefineMapNodelet] even after refreshing the cache: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//librefine_map_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [ERROR] [1596707128.000320047]: The error before refreshing the cache was: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//librefine_map_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [FATAL] [1596707128.000427138]: Failed to load nodelet '/refine_map_nodelet` of type `hdl_graph_slam/RefineMapNodelet` to manager `velodyne_nodelet_manager'
> [ERROR] [1596707128.026613695]: Failed to load nodelet [/prefiltering_nodelet] of type [hdl_graph_slam/PrefilteringNodelet] even after refreshing the cache: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libprefiltering_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [ERROR] [1596707128.026666448]: The error before refreshing the cache was: Failed to load library /home/hurricanezj/catkin_ws/devel/lib//libprefiltering_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libvtkFiltersModeling-6.2.so.6.2: cannot open shared object file: No such file or directory)
> [FATAL] [1596707128.026863350]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
> [scan_matching_odometry_nodelet-5] process has died [pid 2489, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/scan_matching_odometry_nodelet-5.log].
> log file: /home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/scan_matching_odometry_nodelet-5*.log
> [floor_detection_nodelet-6] process has died [pid 2490, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/FloorDetectionNodelet velodyne_nodelet_manager __name:=floor_detection_nodelet __log:=/home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/floor_detection_nodelet-6.log].
> log file: /home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/floor_detection_nodelet-6*.log
> [hdl_graph_slam_nodelet-7] process has died [pid 2491, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/hdl_graph_slam_nodelet-7.log].
> log file: /home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/hdl_graph_slam_nodelet-7*.log
> [refine_map_nodelet-8] process has died [pid 2493, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/RefineMapNodelet velodyne_nodelet_manager __name:=refine_map_nodelet __log:=/home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/refine_map_nodelet-8.log].
> log file: /home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/refine_map_nodelet-8*.log
> [prefiltering_nodelet-4] process has died [pid 2488, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager /velodyne_points:=/rslidar_points __name:=prefiltering_nodelet __log:=/home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/prefiltering_nodelet-4.log].
> log file: /home/hurricanezj/.ros/log/8a983452-d7c9-11ea-a84b-9cb6d0effb2d/prefiltering_nodelet-4*.log

FishInWave, 2020-08-17:
I have found that there is something wrong with my pcl lib, and this error is not related with hdl-graph-slam, so I close this issue.

closed 2020-08-17 by FishInWave

GigaFlopsis, 2020-11-11:
I have the same problem. How you fixed it?

numb0824, 2021-08-25:
Hi, i have the same problem.Could you plz tell me that how to fix this error?