koide3/hdl_graph_slam #142: ERROR: cannot launch node of type [msf_updates/pose_sensor]: msf_updates

Johntec, 2020-09-03:
ystem: ubuntu 18.04
Hi @koide3 ,
I have the same problem which was closed in the issue #138 
ros: melodic
problem:
ERROR: cannot launch node of type [msf_updates/pose_sensor]: msf_updates
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/cakgod/hdl_graph_slam/src
ROS path [2]=/home/cakgod/ndt_omp/src
ROS path [3]=/opt/ros/melodic/share
is there any new libraries to be installed ?

Johntec, 2020-10-22:
Hi all, does anyone has an idea how to solve this issue?

h-wata, 2020-12-04:
In my case, this issue is solved as below:
OS: Ubuntu 18.04
ROS: melodic

```
cd ~/ROS_WORKSPACE/src/
git clone https://github.com/catkin/catkin_simple.git
git clone https://github.com/ethz-asl/glog_catkin.git
git clone https://github.com/ethz-asl/ethzasl_msf.git
catkin build
```

I hope this will help you.

Johntec, 2021-01-02:
Thank you very much @h-wata , your solution has solved the problem.

closed 2021-01-02 by Johntec

donkorechi, 2021-10-05:
This is what worked for me.

```
sudo apt install libgoogle-glog-dev

cd ~/ROS_WORKSPACE/src/
git clone https://github.com/ethz-asl/ethzasl_msf.git
cd ..
catkin_make -DCMAKE_BUILD_TYPE=Release

```