koide3/hdl_graph_slam #143: Mapping without floor

jhjun724, 2020-09-08:
Hi. First of all, thank you for your packages.
I'm trying to save the map without floor.
It was really easy to save the floor only, but I have no idea about mapping without floor.
I edited 'extract.setNegative();' in "floor_detection_nodelet.cpp" and tried other things.
There was some changes but the result was bad.
Is there any solution to solve this problem?
Thank you.

closed 2021-03-04 by jhjun724

k10DAI, 2023-07-03:
@jhjun724 Can you please explain how you solved this? I also want to save the map (outdoor env.) without the floor points. I set `enable_floor_detection` to false but doesn't seem to help.

yigitboracagiran, 2023-08-09:
Have you solved it yet? I'm facing the same problem.