koide3/hdl_graph_slam #145: working fine but [pcl::NormalEstimation::compute] input_ is empt! keep print out

ipinyulun, 2020-09-16:
Hi,

i could have the mapping working. 
however there is error 
`[pcl::NormalEstimation::compute] input_ is empt!` keep print out
as per image below.. everything else i guess working fine since i could see the map cloud(white)) and the input cloud(rainbow)

[image]

the different is ... 
when i use rosbag recorded from velodyne VLP16, its working fine.. no such error..

but i get the error above when i use sick mrs1104 (4 laser, 7.5 degree vertical)

im aware that this error come from floor_detection_nodelet,
however no matter  i tweak the param, sensor_height, height_clip_range, and floor_pts_thresh...
it wont eliminate that error

the bag sample is here
https://drive.google.com/file/d/1lD3vWHrTpi0Pz300FIPkR5cBRB5mjCtJ/view?usp=sharing

ipinyulun, 2020-09-16:
and here is my launch file

<?xml version="1.0"?>
<launch>
    <param name="/use_sim_time" value="true" />
  
  <arg name="nodelet_manager" default="velodyne_nodelet_manager" />
  <arg name="enable_floor_detection" default="true" />
  <arg name="enable_gps" default="false" />
  <arg name="enable_imu_acc" default="false" />
  <arg name="enable_imu_ori" default="false" />
  <arg name="points_topic" default="/velodyne_points" />

  
  <node pkg="tf" type="static_transform_publisher" name="lidar2base_publisher" args="0 0 0 0 0 0 base_link velodyne 100" />

  
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>

  
  <node pkg="nodelet" type="nodelet" name="prefiltering_nodelet" args="load hdl_graph_slam/PrefilteringNodelet $(arg nodelet_manager)">
    <remap from="/velodyne_points" to="$(arg points_topic)" />
    
    <param name="base_link_frame" value="base_link" />
    
    <param name="use_distance_filter" value="true" />
    <param name="distance_near_thresh" value="0.1" />
    <param name="distance_far_thresh" value="100.0" />
    
    <param name="downsample_method" value="VOXELGRID" />
    <param name="downsample_resolution" value="0.1" />
    
    <param name="outlier_removal_method" value="RADIUS" />
    <param name="statistical_mean_k" value="30" />
    <param name="statistical_stddev" value="1.2" />
    <param name="radius_radius" value="0.5" />
    <param name="radius_min_neighbors" value="2" />
  </node>
<node type="rviz" name="rviz" pkg="rviz" args="-d /home/nvidia/catkin_ws/src/hdl_graph_slam/rviz/hdl_graph_slam.rviz" /> 

  
  <node pkg="nodelet" type="nodelet" name="scan_matching_odometry_nodelet" args="load hdl_graph_slam/ScanMatchingOdometryNodelet $(arg nodelet_manager)">
    <param name="points_topic" value="$(arg points_topic)" />
    <param name="odom_frame_id" value="odom" />
    <param name="keyframe_delta_trans" value="1.0" />
    <param name="keyframe_delta_angle" value="1.0" />
    <param name="keyframe_delta_time" value="1.0" />
    <param name="transform_thresholding" value="true" />
    <param name="max_acceptable_trans" value="1.0" />
    <param name="max_acceptable_angle" value="1.0" />
    <param name="downsample_method" value="NONE" />
    <param name="downsample_resolution" value="0.1" />
    
    <param name="registration_method" value="NDT_OMP" />
    <param name="ndt_resolution" value="1.0" />
    <param name="ndt_num_threads" value="0" />
    <param name="ndt_nn_search_method" value="DIRECT7" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="floor_detection_nodelet" args="load hdl_graph_slam/FloorDetectionNodelet $(arg nodelet_manager)" if="$(arg enable_floor_detection)">
    <param name="points_topic" value="$(arg points_topic)" />

    
        <param name="sensor_height" value="2.0" />
    <param name="height_clip_range" value="1.0" />
    <param name="floor_pts_thresh" value="256" />
  </node>

  
  <node pkg="nodelet" type="nodelet" name="hdl_graph_slam_nodelet" args="load hdl_graph_slam/HdlGraphSlamNodelet $(arg nodelet_manager)">
    <param name="points_topic" value="$(arg points_topic)" />
    
    <param name="map_frame_id" value="map" />
    <param name="odom_frame_id" value="odom" />
    
    
    <param name="g2o_solver_type" value="lm_var_cholmod" />
    <param name="g2o_solver_num_iterations" value="512" />
    
    <param name="enable_gps" value="$(arg enable_gps)" />
    <param name="enable_imu_acceleration" value="$(arg enable_imu_acc)" />
    <param name="enable_imu_orientation" value="$(arg enable_imu_ori)" />
    
    <param name="max_keyframes_per_update" value="10" />
    <param name="keyframe_delta_trans" value="2.0" />
    <param name="keyframe_delta_angle" value="2.0" />
    
    <param name="fix_first_node" value="true"/>
    <param name="fix_first_node_stddev" value="10 10 10 1 1 1"/>
    <param name="fix_first_node_adaptive" value="true"/>
    
    <param name="distance_thresh" value="0.5" />
    <param name="accum_distance_thresh" value="3.0" />
    <param name="min_edge_interval" value="1.0" />
    <param name="fitness_score_thresh" value="0.5" />
    
    <param name="registration_method" value="NDT_OMP" />
    <param name="ndt_resolution" value="1.0" />
    <param name="ndt_num_threads" value="0" />
    <param name="ndt_nn_search_method" value="DIRECT7" />
    
    
    <param name="gps_edge_robust_kernel" value="NONE" />
    <param name="gps_edge_robust_kernel_size" value="1.0" />
    <param name="gps_edge_stddev_xy" value="1000.0" />
    <param name="gps_edge_stddev_z" value="5.0" />
    
    <param name="imu_orientation_edge_robust_kernel" value="NONE" />
    <param name="imu_orientation_edge_stddev" value="1.0" />
    
    <param name="imu_acceleration_edge_robust_kernel" value="NONE" />
    <param name="imu_acceleration_edge_stddev" value="1.0" />
    
    <param name="floor_edge_robust_kernel" value="NONE" />
    <param name="floor_edge_stddev" value="10.0" />
    
    
    <param name="odometry_edge_robust_kernel" value="NONE" />
    <param name="odometry_edge_robust_kernel_size" value="1.0" />
    <param name="loop_closure_edge_robust_kernel" value="Huber" />
    <param name="loop_closure_edge_robust_kernel_size" value="1.0" />
    <param name="use_const_inf_matrix" value="false" />
    <param name="const_stddev_x" value="0.5" />
    <param name="const_stddev_q" value="0.1" />
    <param name="var_gain_a" value="20.0" />
    <param name="min_stddev_x" value="0.1" />
    <param name="max_stddev_x" value="5.0" />
    <param name="min_stddev_q" value="0.05" />
    <param name="max_stddev_q" value="0.2" />
    
    <param name="graph_update_interval" value="1.5" />
    <param name="map_cloud_update_interval" value="3.0" />
    <param name="map_cloud_resolution" value="0.01" />
  </node>

  <node pkg="hdl_graph_slam" type="map2odom_publisher.py" name="map2odom_publisher" />
</launch>

BamsaAhmed, 2023-01-11:
Hi did you solve this issue?

mgrallos, 2023-05-26:
i am encountering the same issue. any workaround?