koide3/hdl_graph_slam #153: velodyne_nodelet_manager

wayneasimo, 2020-10-22:
Hi guys,
when I try to rosbag play xxx.bag, the following problem occured , I don't know how to solve it:

`[FATAL] [1603334627.870802907, 1509348797.533911957]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1603334627.870990905, 1509348797.533911957]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 10560, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/velodyne_nodelet_manager-2.log].
log file: /home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/velodyne_nodelet_manager-2*.log
[scan_matching_odometry_nodelet-4] process has died [pid 10567, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/scan_matching_odometry_nodelet-4.log].
log file: /home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/scan_matching_odometry_nodelet-4*.log
[hdl_graph_slam_nodelet-6] process has died [pid 10574, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/hdl_graph_slam_nodelet-6.log].
log file: /home/wayne/.ros/log/695b3efe-140c-11eb-bc77-7cb27d5f2436/hdl_graph_slam_nodelet-6*.log
`

Johntec, 2020-10-22:
hi @wayneasimo , did you source the package? if not then try $_`source ~/your_catkin/devel/setup.bash`_
the start the launch file.

wayneasimo, 2020-10-22:
> hi @wayneasimo , did you source the package? if not then try $_`source ~/your_catkin/devel/setup.bash`_
> the start the launch file.

hi, I have sourced it bu it didn't work

wayneasimo, 2020-10-22:
> hi @wayneasimo , did you source the package? if not then try $_`source ~/your_catkin/devel/setup.bash`_
> the start the launch file.

and my tf tree has no transform from base_link to odom

Dysonsun, 2021-04-25:
Did you fix it?

numb0824, 2021-08-23:
i have met the same problem,could you have solved it ?