koide3/hdl_graph_slam #168: Mapping result looks sparse and unrecognizable

zxl19, 2020-12-27:
Hi, @koide3 !
Thank you for your excellent work! I used the default configuration in `hdl_graph_slam.launch` in my own task, but the mapping result looks weird. The map is sparse and the objects in the map looks unrecognizable when zoomed in.

[image]
[image]
[image]

I referred to #87 and tried the following two configurations in the prefiltering_nodelet, but it doesn't seem to help.

```xml
<param name="downsample_method" value="VOXELGRID" />
<param name="downsample_method" value="APPROX_VOXELGRID" />
```
I wonder what seems to be the problem.

Best wishes.

zxl19, 2020-12-27:
Reupload screenshots:

link：https://pan.baidu.com/s/1v31Wjuj6cZGwRDqK-fSrMg 
extraction code：mjc2

koide3, 2021-01-02:
- What voxel resolution did you use when saving the cloud?
- Did you use GPS? When a map is converted into the UTM coordinate, point coordinates tend to be very large and the PCD file format, which uses single-precision floats, may not be able to properly handle such points.

koide3, 2021-01-04:
> I used 0.01 and 0.05 respectively and it all got basically the same results. Also, I would like to know what your recommendation of the downsampling method and resolution for dense mapping is.

I think 0.1 or 0.05 is a good choice. To small resolution would not be very effective.

> Yes, I did use GPS. It seems that this is what caused the problem. Are there any optional ways to solve this? Can I save the map in UTM orientation but in relative coordinates? Or should I modify the I/O part of the code?

You can set ```utm``` flag to false when calling the save service. Then, the map point coordinates will be relative position w.r.t. the very first frame. After saving the map, a file named ```*.utm``` is created that describes the UTM coordinate of the origin of the map. You can add these values to each point to convert the map point coordinates into the UTM coordinate system.

https://github.com/koide3/hdl_graph_slam/blob/b1d9b51020e705a5c33bd9e1b1ec523781f93782/srv/SaveMap.srv#L1-L5

zxl19, 2021-01-04:
Hi @koide3 . I'll try that, thanks!

closed 2022-04-15 by zxl19

Thomas-Merlet, 2022-12-02:
I've runned into the same issue and found out it's actually related to the size of the UTM offset applied on pcl::PointXYZ.
More info can be found here: https://stackoverflow.com/questions/57465898/precision-getting-lost-when-transforming-the-pcl-pointcloud

I'm converting the pcd directly to ENU MRGS coordinates to avoid that.