koide3/hdl_graph_slam #169: LiDAR coordinate direction?

zxl19, 2020-12-27:
Hi @koide3 !

Thank you for your work! I'm trying to transform my GPS measurement to my LiDAR frame. I've calibrated the extrinsic between my GPS device's coordinate system and LiDAR coordinate system. However, I wonder how the LiDAR coordinate is defined in hdl_graph_slam. Is it defined separately or just uses the original LiDAR coordinate as the LiDAR is mounted on the vehicle?

e.g. the Velodyne HDL64 coordinate direction on the vehicle: x (rear), y (right), z (up). If I use pointcloud from this device, how is the LiDAR coordinate system defined in hdl_graph_slam?

Sincerely.

koide3, 2021-01-02:
Hi @zxl19 ,

hdl_graph_slam assumes that LiDAR and GPS have the same coordinate systems because GPS measurement errors are typically larger than the displacement between LiDAR and GPS. During mapping, it follows the ROS coordinate system convention: X-Y-Z = front-left-up.

GLKode, 2021-01-02:
Hi @zxl19 , how did you calibrate your Velodyne with your IMU ? can you share the steps please?

zxl19, 2021-01-02:
Hi @koide3 !

> hdl_graph_slam assumes that LiDAR and GPS have the same coordinate systems because GPS measurement errors are typically larger than the displacement between LiDAR and GPS.

I'm using a GPS/IMU device made by Novatel and the precision looks pretty good. I'm using GPS mainly to correct the vertical drift, so I think considering the extrinsic may help to increase mapping accuracy. Also, I decreased `gps_edge_stddev_z` value in the launch file. 

```xml
<param name="gps_edge_stddev_z" value="5.0" />
```

But I found the map may tilt (in the vehicle's roll direction) a lot in the beginning and before it gets corrected as the vehicle turns. I wonder what caused the problem and what parameters in the launch file can be tuned in my case for better performance.

> During mapping, it follows the ROS coordinate system convention: X-Y-Z = front-left-up.

According to the LiDAR's user manual, the manufacturer defined the coordinate system of the LiDAR. However, when a LiDAR is mounted on a vehicle it may not follow the ROS convention. So I wonder what will happen if I use the data acquired this way. This is really important for me because I need to make sure which extrinsic should I calibrate, the extrinsic between GPS and my LiDAR coordinate system, or the extrinsic between GPS and ROS conventional coordinate system?

Best wishes.

zxl19, 2021-01-02:
Hi @GLKode !

I'm not sure if your IMU device returns the position, but there are a couple of open-source tools I found that may help you:

[APRIL-ZJU/lidar_IMU_calib](https://github.com/APRIL-ZJU/lidar_IMU_calib)
[ethz-asl/lidar_align](https://github.com/ethz-asl/lidar_align)
[FENGChenxi0823/SensorCalibration](https://github.com/FENGChenxi0823/SensorCalibration)
[apollo_lidar_imu_calibration_guide.md](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/apollo_lidar_imu_calibration_guide.md)

I'm using a GPS/IMU device that not only provides orientation but also latitude, longitude, and altitude. So I used the hand-eye calibration method to calibrate the extrinsic. The code and steps can be found here: [zxl19/Hand_Eye_Extrinsic_Calibration](https://github.com/zxl19/Hand_Eye_Extrinsic_Calibration)

Sincerely.

GLKode, 2021-01-04:
Hi @zxl19 , thank you very much for the information, they are really helpful.
One more question though, when including gps/imu into the slam algorithm, you get the map and the trajectory at the end,  referring to #171 , do you know where can we get the trajectory and in which coordinate frame would it be? 
Bests,

zxl19, 2021-01-04:
Hi @GLKode , you can get odometry data from topic `/odom`. However, the pose estimates may be changed in the pose graph later. Also, you can try calling the rosservice `/hdl_graph_slam/dump` after running your bag, there is a `.g2o` file containing the poses and other constraints in the pose graph. I haven't tried this with GPS yet, but if you only use LiDAR, the pose estimates are w.r.t. the initial position.

GLKode, 2021-01-04:
Thank you @zxl19 Will try that :)

zxl19, 2021-01-04:
Yeah, no problem. Good luck with that. @GLKode

closed 2022-04-15 by zxl19