koide3/hdl_graph_slam #170: More Dense and Detailed Map

ManChrys, 2020-12-30:
Hello @koide3 love your work!!!

First of all Merry Christmas and i wish health and happiness to all!!!!

I want to make my map with more density and more details. I saw your suggestions on
#87 and #38 issues but i dont see big differences!

Im using robosense lidar16.

I run the bag with : 

rosparam set use_sim_time true
rosbag play -r 0.5 82.bag

my launch file is :

[launch.txt](https://github.com/koide3/hdl_graph_slam/files/5753423/launch.txt)

my bag file is 3.7 GB because its an outdoor part of Santorini Island, thats why im sending it with wetransfer. 

bag file : https://we.tl/t-c969H8hlPK

I capture the whole island and almost the whole Cyclades Islands and i need good results.

Thanks in advance!

koide3, 2021-01-02:
Hi @ManChrys ,

Thanks for your greetings, and Happy new year!

I tweaked your launch file a bit so it yields a better map. But, this kind of online SLAM is not very suitable for generating a dense map, and you may need to consider another offline mapping systems for your purpose.

[launch.txt](https://github.com/koide3/hdl_graph_slam/files/5760339/launch.txt)

ManChrys, 2021-01-02:
Hi @koide3 ,

Happy new year my friend!! Thank you for the response. I'll test it on Monday and i'll inform you.

Conserning the last thing you mentioned, do you have any suggestions about offline mapping systems ?

Thanks in advance.

koide3, 2021-01-04:
Our ```interactive_slam``` package provides dense mapping capability but in a closed source version. Please see: https://github.com/SMRT-AIST/interactive_slam/wiki/Closed-source-version

ManChrys, 2021-01-04:
wow it seems that your method has very very good results. So how can i use it ?

ManChrys, 2021-01-05:
@koide3 ok i see what do you mean but the thing is that i was recording pcap files, i didnt do live slam. 

after that i  1) rosbag record /rslidar_points, 2) rosbag play bagfile.bag and then i use interactive_slam.

Thats the order i process the data.

Thanks in advance.

koide3, 2021-01-09:
Because interactive_slam uses an output of hdl_graph_slam, the process order would be:
1. play pcap (or saved rosbag)
2. create a map with hdl_graph_slam
3. save the map with "hdl_graph_slam/dump" service
4. use interactive_slam

ManChrys, 2021-01-09:
Yes that's exactly the way I'm working it . Maybe I didn't describe it very good . Sorry if I wasn't clear. I'm playing pcap which I recorded it and then I follow exactly the same steps which you described . After interactive slam the map isn't dense though. I don't know what I'm doing wrong . I see all of your issues and follow your advices but still something is missing . I saw your wiki link which you send me earlier and it seems that you have nice results . Although my map isn't so dense and I don't know what I am doing wrong 😞

koide3, 2021-01-20:
The dense mapping function is included in only the closed source version of ```interactive_slam```. If you really need it, you may ask [my institute](https://unit.aist.go.jp/intpro/index_en.html), but I'm not really sure if they can ship it to foreign countries...

ManChrys, 2021-01-22:
@koide3 
ok my friend thank you for your advises. Have a nice day.

closed 2021-01-22 by ManChrys