koide3/hdl_graph_slam #171: trajectory coordiante frame

GLKode, 2021-01-02:
Hi @koide3 ,

are the messages published on /odom topic the resulted trajectory of hd graph slam? and what coordinate frame are they?

koide3, 2021-01-02:
```/odom``` topic publishes "odometry" data without loop and floor constraints. It's origin is the very first frame of the LiDAR data, and may have some estimation drift.

GLKode, 2021-01-02:
Thanks @koide3 for the reply,so on which topic the end trajectory including all constraints such as loops and floor points ia published?

koide3, 2021-01-04:
The corrected trajectory is not be published. But, you can use tf lookup to find the latest sensor pose in the map coordinate.

GLKode, 2021-01-04:
thanks @koide3 for the reply. so even if I collect the poses, the jumps in the trajectory would correct the error in the previous poses with the loop closure and change the positions of the poses. do you know how a function can be implemented to generate a trajectory at the end when the map is complete, considering all contains, gps/imu, floor points, loops and lidar/odom ? so that at the end you have a pcd file and a trajectory file. 

you use the markers to create a trajectory in rviz, can this be further developed for this purpose ?

koide3, 2021-01-04:
To obtain the final estimation result of the trajectory, you can use the ```dump``` service that saves the final pose graph with all constraints. Note that the pose graph has only keyframe poses, and thus poses of some frames would be missing. To recover the full sensor trajectory, you would need to do some smoothing on the optimized keyframe poses.

GLKode, 2021-01-04:
Thank you @koide3 will try that, what about the hdl_localization how can i get the final trajectory there? can you give me a hint about that also please?

koide3, 2021-01-09:
Since hdl_localization publishes map aligned poses, you can just subscribe to ```/odom``` topic to get the final trajectory.

GLKode, 2021-01-12:
Thank you @koide3 for the information :)

closed 2021-01-12 by GLKode