koide3/hdl_graph_slam #173: Loop detection deadlock

Johntec, 2021-01-12:
Hello @koide3 

I have encountered problems with loop detection, in many instances when loop matching score is high, while the bag file is almost at the end the loop detection process is stuck in a point and in the same area searching for loops 10s of times. Is there a way to solve this issue?

Thank you for the great work.

ikhann, 2021-01-15:
Hello, could you explain what do "xxx_edge_robust_kernel" and "xxx_edge_robust_kernel_size"  mean in pose graph optimization algorithm? Thanks!

koide3, 2021-01-20:
Robust kernels are helpful when inputs are noisy and contain some outliers. It would be good to use it for loop constraints that may have wrong loops. Maybe it is also good to apply it for GPS constraints when GPS data can be inaccurate.