koide3/hdl_graph_slam #175: Remove moving objects from the map

ikhann, 2021-01-15:
Hello, 

Do you have any ideas about how to detect and remove moving objects (pedestrian crossing the road, moving cars) from the generated 3d map or from each lidar scan on the prefiltering phase?

Thank you for advising me!

koide3, 2021-01-20:
Probably, you can do some segmentation to find moving objects and then remove them from the input cloud before feeding it to the prefiltering nodelet

ikhann, 2022-01-20:
Hi @koide3!
I have added a support of hdl_graph_slam format for removing dynamic points from map using [Removert](https://github.com/irapkaist/removert). It works great!

Pull request https://github.com/irapkaist/removert/pull/15

koide3, 2022-01-22:
Hi @ikhann ,
I'm so happy to see your attempt to connect hdl_graph_slam with Removert. Thanks a lot for letting me know :smile:

closed 2022-01-23 by ikhann