koide3/hdl_graph_slam #180: Accuracy of the map

ikhann, 2021-02-11:
Hello!
Thanks for this awesome framework!

Could you advise please.
How can achieve such accuracy of the outdoor map to have a single layer of ground (one surface) without doubled ground or "thick film" of the ground?

Thank you!

koide3, 2021-02-16:
I think enabling floor constraint helps in resolving the doubled ground issue.

ikhann, 2021-02-16:
@koide3 Thank you for your reply!

Is it correct to enable "floor detection" option for an outdoor map with a road slope with ascents and descents?

koide3, 2021-02-17:
The floor plane correction assumes that the environment has a flat floor. If you need to consider ascents/descents or multiple floors, please consider using [interactive_slam](https://github.com/SMRT-AIST/interactive_slam) that enables manually inserting plane constraints to the map.

ikhann, 2021-02-17:
I've already started using it. 
Thanks a lot!

closed 2021-02-17 by ikhann