koide3/hdl_graph_slam #182: UTM origin coordinate precision loss

ksuszka, 2021-02-13:
When a final map is saved with utm: false parameter, then UTM origin is substracted from all points and it is saved to accompanying .utm file. However precision of serialized coordinates is limited and it can cause a few meters displacement.

In my case, value "5572094.155411" was saved as "5.57209e+06".

koide3, 2021-02-16:
Thanks for reporting the issue. I fixed the text format so the precision is preserved. It'll be merged into the main branch soon.

ksuszka, 2021-02-28:
I just noticed that there are two separate functions affected by this formatting issue. I've create PR https://github.com/koide3/hdl_graph_slam/pull/185 with your fix applied to the second function.