koide3/hdl_graph_slam #204: How can I use GPU?

OmerKonan, 2021-06-14:
Hello,
I saw #ifdef USE_VGICP_CUDA  lines for using FAST_GICP method with cuda.
Can you explain how can I define USE_VGICP_CUDA?

Thanks

koide3, 2021-06-15:
GPU-based features can be turned on by passing ```-DBUILD_VGICP_CUDA=ON``` to catkin_make: 
```
catkin_make -DCMAKE_BUILD_TYPE=Release -DBUILD_VGICP_CUDA=ON
```

OmerKonan, 2021-06-15:
Hello,
Thank you for quick answer. I have tried twice but still got the same error like that:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "fast_vgicp_cuda" in directory

Do you have any solution?

koide3, 2021-06-22:
Maybe cmake failed to find cublas libracy. Can you check if CUDA-related libraries are properly installed?

closed 2021-06-25 by OmerKonan