koide3/hdl_graph_slam #211: Spiral-like pointclouds in some places

awesomebytes, 2021-07-28:
Hello. Thank you very much for this interesting package. I've been playing with it and I found that I get some interestingly looking pointclouds where there are some 'giant spirals' of points, as if a set of pointclouds were attempted to be merged in a 360 degrees fashion.

See screenshot attached (from interactive slam, which I'm also checking out). Any idea what may be creating these? (and how to avoid it?) This map was created with GPS data, I also tried with IMU and GPS and the same happened. Without GPS & IMU the map has wrong loop closures so I'm not really taking it into account.

[image]

koide3, 2021-07-29:
Hmm, I've not seen this problem, and I'm not really sure what is the cause (maybe corrupted IMU or loop constraints are inserted?)... If you could share your graph dump data, it would be helpful to dig into the problem.

awesomebytes, 2021-07-29:
Seems like it has something to do with using GPS contraints. Without GPS I don't see these happening. However, the graph is pretty much wrong and never does loop closures. I need to use interactive_slam to introduce loop closures.

I don't consider this issue serious so I don't want to waste your time (and my current dump folder is about 6GB, a bit annoying to upload). Thanks a lot for replying that quickly!

closed 2021-07-29 by awesomebytes