koide3/hdl_graph_slam #213: msf data too old

wongkenchiang97, 2021-08-08:
Hi, I'm running "_hdl_graph_slam_400.launch_" with my bagfile. Sometimes the SLAM optimized results was good, but sometimes was bad.

In launch file, my "_enable_imu__" options are "_false_".
In bag file I have used "_--clock_" argument and didn't remap imu msg.
[image]

Has anyone also encounter this issue?
Could this error affect the optimization results?

donkorechi, 2021-10-05:
sudo apt install libgoogle-glog-dev

cd ~/ROS_WORKSPACE/src/
git clone https://github.com/ethz-asl/ethzasl_msf.git
cd ..
catkin_make -DCMAKE_BUILD_TYPE=Release

don-thomas, 2021-10-07:
I was able to fix this problem by doing the following:

```
sudo apt install libgoogle-glog-dev 
cd ~/ros_workplace/src
git clone https://github.com/ethz-asl/ethzasl_msf.git
cd .. && catkin_make -DCMAKE_BUILD_TYPE=Release
```