koide3/hdl_graph_slam #215: final optimized odometry for comparing accuracy of slam

wakasu, 2021-09-01:
Hey,

I'm new to slam so sorry if its an obvious question

I am trying to save the final optimized odometry of in "timestamp tx ty tz qx qy qz qw" format

Any help will be appreciated.
Regards

Remi-Tortue, 2022-05-17:
Hey,

I have the same issue. Did you find a solution ?

njustCSwy, 2024-01-12:
you can add this code into file hdl_graph_slam_nodelet.cpp at method dump_service().
    std::ofstream foutTUM;
    foutTUM.open("/home/indigo/catkin_hdl/src/tum/tum.txt");
    foutTUM << std::fixed;
    for(int i = 0; i < keyframes.size(); i++)
    {
      Eigen::Vector3d translation = keyframes[i]->odom.translation();
      Eigen::Quaterniond rotation(keyframes[i]->odom.rotation());
      //save the traj
      foutTUM << 
      std::setprecision(6) << keyframes[i]->stamp.toSec() << " " 
      << std::setprecision(9) << translation.x() << " " 
      << translation.y() << " " 
      << translation.z() << " " 
      << rotation.x() << " " 
      << rotation.y() << " " 
      << rotation.z() << " " 
      << rotation.w() << std::endl;
    }
    foutTUM.close();

LIYVang, 2024-04-06:
> 您可以将此代码添加到方法 dump_service（） 中的文件hdl_graph_slam_nodelet.cpp中。 std：：ofstream 主输出TUM; foutTUM.open（“/home/indigo/catkin_hdl/src/tum/tum.txt”）; main outTUM << std：：fixed; for（int i = 0; i < keyframes.size（）; i++） { Eigen：：Vector3d translation = keyframes[i]->odom.translation（）;特征：：四元数旋转（keyframes[i]->odom.rotation（））; //保存 traj foutTUM << std：：setprecision（6） << 关键帧 [i]->stamp.toSec（） << “ ” << std：：setprecision（9） << translation.x（） << “ ” << translation.y（） << “ ” << translation.z（） << “ ” << rotation.x（） << “ ” << rotation.y（） << “ ” << rotation.z（） << “ ” ”<< rotation.w（） << std：：endl; } main wayTUM.close（）;

你好，我在尝试后发现轨迹并不能保存下来