koide3/hdl_graph_slam #219: 3D Navigation? or any options for generating 2D map from 3D map ?

robertsenps, 2021-10-05:
Hi @koide3  ,

First of all, I would like to say thank you to your great 3d pipeline packages. I have used your package for almost 3 months. I did tune so many parameters including the scan_matching parameters and hdl_graph_slam. I successfully got a very decent map. 

I want to use your 3d map also with your hdl localization to make the robot navigate in 3-dimensional space. Because there is no stable 3d navigation pkg on the internet, I decided to use 2d navigation instead. I try to generate a 2d map from 3d map using octomap. But I have a problem with the 2d map generation. I pass the whole 3d pointcloud to the octomap server and set the raytracing to true. I set the raytracing to true because I want to get an output map that is very similar to the 2d mapping like gmapping. The problem is every time the graph is changed (i.e. loop closing), the 2d map on octomap won't be updated according to where the aligned 3d map is located. So the 2d and 3d map is unaligned each other. 

I also tried using individual samples from raw lidar to build an octomap representation, but it's the same.

My questions are:
1. Is there any 3d navigation package that is suitable with hdl_graph_slam ?
2. If there's no 3d navigation pkg, what should I do to get 2d map that is also updated when 3d map is updated/optimized ?

I'm waiting for your reply. Any help from you will be very helpful for me and the community.

Thanks :)

koide3, 2021-10-06:
> Is there any 3d navigation package that is suitable with hdl_graph_slam ?

In my opinion, 3D navigation on ROS is not very matured, and I usually use the standard 2D navigation stack with a 2D (static) map generated offline. In my case, I didn't need a sophisticated navigation and used a very naive 2D map generation (extracting points in a certain height range):

https://github.com/koide3/pointcloud_to_2dmap

> If there's no 3d navigation pkg, what should I do to get 2d map that is also updated when 3d map is updated/optimized ?

If you want to navigate the robot while dynamically creating the map, yes, you need to re-create the 2D map every time the 3D map is updated.

arjunskumar, 2023-01-19:
@dionesiusap have you manged to implement dynamic 2d map?

joontack-han, 2023-04-26:
@robertsenps @dionesiusap @arjunskumar Well have anybody got solution of 2d mapping? I was going to use @koide3's pcdto2dmap but changed direction to octomap since I couldn't run those properly.

joontack-han, 2023-04-26:
@arjunskumar Have u tried octomap or pointcloud_to_2dmap mentioned?

20-wash, 2024-12-13:
Has anyone found the solution for realtimr  conversion from 3d map to 2d map? Can anyone post the link and the documentation?