koide3/hdl_graph_slam #224: hdl_graph_slam hdl_graph_slam_xxx.launch FAILED

promistrio, 2021-12-07:
As in previous topics, I faced a similar problem. Ubuntu 20, ros noetic. I've tried all the tips from #4. I was running from under docker. I built h2o from sources. But the problem remained.

```

[FATAL] [1638908975.504756434]: Failed to load nodelet '/hdl_graph_slam_nodelet` of type `hdl_graph_slam/HdlGraphSlamNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1638908975.504868233]: Failed to load nodelet '/prefiltering_nodelet` of type `hdl_graph_slam/PrefilteringNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1638908975.505105364]: Failed to load nodelet '/floor_detection_nodelet` of type `hdl_graph_slam/FloorDetectionNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1638908975.505278242]: Failed to load nodelet '/scan_matching_odometry_nodelet` of type `hdl_graph_slam/ScanMatchingOdometryNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-3] process has died [pid 11634, exit code -4, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/velodyne_nodelet_manager-3.log].
log file: /home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/velodyne_nodelet_manager-3*.log
[prefiltering_nodelet-4] process has died [pid 11635, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load hdl_graph_slam/PrefilteringNodelet velodyne_nodelet_manager /velodyne_points:=/velodyne_points __name:=prefiltering_nodelet __log:=/home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/prefiltering_nodelet-4.log].
log file: /home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/prefiltering_nodelet-4*.log
[scan_matching_odometry_nodelet-5] process has died [pid 11641, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load hdl_graph_slam/ScanMatchingOdometryNodelet velodyne_nodelet_manager __name:=scan_matching_odometry_nodelet __log:=/home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/scan_matching_odometry_nodelet-5.log].
log file: /home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/scan_matching_odometry_nodelet-5*.log
[floor_detection_nodelet-6] process has died [pid 11642, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load hdl_graph_slam/FloorDetectionNodelet velodyne_nodelet_manager __name:=floor_detection_nodelet __log:=/home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/floor_detection_nodelet-6.log].
log file: /home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/floor_detection_nodelet-6*.log
[hdl_graph_slam_nodelet-7] process has died [pid 11647, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load hdl_graph_slam/HdlGraphSlamNodelet velodyne_nodelet_manager __name:=hdl_graph_slam_nodelet __log:=/home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/hdl_graph_slam_nodelet-7.log].
log file: /home/vlad/.ros/log/63756792-579c-11ec-a5bc-0fdb42f8c95c/hdl_graph_slam_nodelet-7*.log

```

What could be the problem?

zhouyong1234, 2022-07-01:
hdl_graph_slam 的 CMakeLists 文件添加 add_compile_options(-std=c++11)

Shangareer, 2022-07-08:
> 

It didn't work for me. The environment of my computer is Ubuntu18.04, ros-melodic, pcl-1.8.

LopezBejarAlvaro, 2022-11-21:
Hi @Shangareer, I am also using ros-melodic and pcl-1.8, where you able to solve this problem? Thanks!