koide3/hdl_graph_slam #226: Some issues while running imu example

DaddyWesker, 2022-01-25:
Hello and thanks for your code.

I've been trying to run some examples on your code for a while and got some strange results with imu example. I've been using kitti sequence which I've transformed into rosbag using kitti2bag python lib.
1. When i'm launching  `roslaunch hdl_graph_slam hdl_graph_slam_imu.launch` I'm getting 
```
ERROR: cannot launch node of type [msf_updates/pose_sensor]: msf_updates
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/daddywesker/catkin_ws/src
ROS path [2]=/opt/ros/noetic/share
```
Don't know if this is fine, though node won't abort.
2. When I'm launching then rosbag with 
`rosbag play --clock  kitti_2011_09_29_drive_0071_synced.bag /kitti/velo/pointcloud:=/velodyne_points /kitti/oxts/imu:=/imu/data`
I'm seeing that for some reason after first iter of map creation it won't update at all. I've recorded a video of what i'm doing and what i'm getting (no comments). 

https://drive.google.com/file/d/185nHqUYEN4Jls-flWDYpEYPDkF7asC3L/view?usp=sharing

SO, am i doing something wrong or does it supposed to be like this?

I'm using linux Mint 20.1

koide3, 2022-01-27:
To use the IMU feature, you need to install [ethzasl_sensor_fusion](http://wiki.ros.org/ethzasl_sensor_fusion). As it's apt package is no longer available for ROS noetic, you would need to build it by yourself.

DaddyWesker, 2022-01-27:
Hm, so installation procedure in `4. **Installation**` no longer availiable for noetic? So, i should build it using cmake i guess? Do i need to perform any other actions after installation?

DaddyWesker, 2022-01-31:
Alright, I have install this package using installation procedure and nothing has changed.

yigitboracagiran, 2023-08-11:
> To use the IMU feature, you need to install [ethzasl_sensor_fusion](http://wiki.ros.org/ethzasl_sensor_fusion). As it's apt package is no longer available for ROS noetic, you would need to build it by yourself.

Can you add this to readme.md please. If anyone search this solution like me can find it faster.