koide3/hdl_graph_slam #232: No SLAM output when using RoboSense bpearl LIDAR

ewan-pugh, 2022-03-03:
I am trying to run hdl_graph_slam with my RS BPearl LIDAR. I have modified the hdl_graph_slam_501 launch file to match the LIDAR's parameters but I am not getting any output. If anyone could help that would be great.
My hdl_graph_slam launch file:
[hdl_graph_slam_501_rs.txt](https://github.com/koide3/hdl_graph_slam/files/8179417/hdl_graph_slam_501_rs.txt)

My RS lidar config file:
[config.txt](https://github.com/koide3/hdl_graph_slam/files/8179431/config.txt)

RVIZ output:
[image]

arjunskumar, 2022-03-10:
@ewan-pugh because  this package expecting the data in velodyne format, the easy way is to convert rslidar data to velodyne data and use the launch file provided with the package

ewan-pugh, 2022-03-10:
@arjunskumar okay thanks for your reply

closed 2022-03-10 by ewan-pugh