koide3/hdl_graph_slam #233: Slam with GPS output

lhakim85, 2022-03-04:
Hi,
We successfully launch the slam with GPS,but how to validate the output? What is the different between slam with GPS and without GPS?

lhakim85, 2022-06-01:
> Hi @lhakim85 ,
> 
> You may expect that GPS must help to state estimation basically. If you increase the GPS accuracy which is not represent the real accuracy form launch file, for example: your GPS sensor has 1 meter error but pass the GPS deviation 0.1 meter, we expect that created map must be worse.
> 
> GPS deviation parameter:
> 
> https://github.com/koide3/hdl_graph_slam/blob/40fd395bdfdea2ae682ae3d01f7e78d528dcadb9/launch/hdl_graph_slam_400.launch#L124

thanks @mertyavuz41 for the reply.
For the value="20.0", this parameter refer to meter/cm??

mertyavuz41, 2022-06-01:
@lhakim85 , it is in meter.