koide3/hdl_graph_slam #237: process has died, exit code -9

AleksandrKashirin, 2022-07-26:
Hello, I am trying to run the algorithm in my own data. Unfortunately, I am receiving this error after a while of processing the data:

[velodyne_nodelet_manager-2] process has died [pid 28341, exit code -9, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/aleksandr/.ros/log/be799298-0cf6-11ed-aca3-c5d0d9103af3/velodyne_nodelet_manager-2.log].
log file: /home/aleksandr/.ros/log/be799298-0cf6-11ed-aca3-c5d0d9103af3/velodyne_nodelet_manager-2*.log

AleksandrKashirin, 2022-07-28:
Update: I  have partially solved this problem by reducing map resolution.
File: `hdl_graph_slam_400.launch` or any other launch file you are working with.
parameter to be changed: `<param name="map_cloud_resolution" value="0.25" />` default value is 0.05. try to increase the value if you faced this problem.
Good luck!

2023swk, 2023-04-21:
I meet the same problems now,but I do not have idea to solve  it.