koide3/hdl_graph_slam #245: Is it possible to keep pointcloud other channel data when save keyframe?

flymaxty, 2022-10-14:
Hi all,
We are using [ouster](ouster.com) lidar with hdl_graph_slam, ouster lidar has additional data(such as ambient) on pointcloud msg, but looks like hdl_graph_slam only keep xyz value on pointcloud. 
Is it possible to keep ambient/reflectivity data when build 3d map, or what should I do to achieve this?

[image]

koide3, 2022-10-14:
If you need only ambient, you can consider modifying the prefiltering nodelet so that it stores ambient values to the intensity channel.

https://github.com/koide3/hdl_graph_slam/blob/46538c737df56cf9b527960a2b3819583d1ff40c/apps/prefiltering_nodelet.cpp#L106

flymaxty, 2022-10-15:
> If you need only ambient, you can consider modifying the prefiltering nodelet so that it stores ambient values to the intensity channel.
> 
> https://github.com/koide3/hdl_graph_slam/blob/46538c737df56cf9b527960a2b3819583d1ff40c/apps/prefiltering_nodelet.cpp#L106

Thanks for your help! I will try it next week :)