koide3/hdl_graph_slam #251: How to use my own odometry

SaltUhey, 2023-01-12:
Hey, I want to use **my own odometry** in your hdl_graph_slam.

According to what was stated in #74,  I have to **remove scan_matching_odometry_nodelet** and **publish my own odometry**  to use my own `/odom` topic.
I tried that.

However, maybe map2odom publisher does not seem to be working well in my case.

When execute `rosbag play <my ros begfile name>.bag -r 0.8 --clock0` in another terminal, an error occurred as shown below
```
[ WARN] [1673493065.274141593, 1672805529.788835466]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame odom at time 1672805529.788836 according to authority unknown_publisher
```
in terminal which execute `rviz -d hdl_graph_slam.rviz`.

I can somewhat make a map, but even after typing `rosrun tf tf_echo map odom` , the following continues to be output.
```
- Translation: [-0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
in RPY (radian) [0.000, -0.000, 0.000]
in RPY (degree) [0.000, -0.000, 0.000]
```

And the `rqt_graph` , `tf tree` and distorted map(using only odometry, scan matching is not working. ) are shown below.

Appreciate for your help.

[image]
[image]
[image]
180-706ed883-10d8-4ed2-a7cf-166ff8620a29.png)

SaltUhey, 2023-01-12:
Sorry, I was mistaken.
I checked the contents of "map2odom_publisher.py", I found that typing `rosrun tf tf_echo map odom` and getting the following output is the correct behavior.
```
- Translation: [-0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
in RPY (radian) [0.000, -0.000, 0.000]
in RPY (degree) [0.000, -0.000, 0.000]
```

However, this method accumulates errors because only wheel odometry is used for mapping.

I would like to use the wheel odometry from my robot as the initial position for scan matching.

I believe that I can use the wheel odometry as the initial position for scan matching by using the optional arguments in the launch file, but I don't know how to use it.
I think it has something to do with the following that exists in each launch file.
 ```
<arg name="enable_robot_odometry_init_guess" default="false" />
<arg name="robot_odom_frame_id" default="robot_odom" />
```
I don't understand what `robot_odom_frame_id` corresponds to.
If you could tell me what it is, I would appreciate it.

rokusottervanger, 2023-11-16:
Same question here. I am trying to set `enable_robot_odometry_init_guess` to `true` and I set the `robot_odom_frame_id` to `odom` (whereas the output of the lidar odometry nodelet is `icp_odom`). However, I'm repeatedly getting this error:
```
[ WARN] [1700147569.776347216, 1699979736.812924720]: failed to look up transform between base_link and odom
```