koide3/hdl_graph_slam #260: please help. hdl_graph_slam launch nodelet issue

lidarmansiwon, 2023-07-18:
hi, i using this package in jetson orin and noetic, 20.04
so when i launch "roslaunch hdl_graph_slam hdl_graph_slam.launch 
I got that error msg. I need your help. please help me,,,

[image]

closed 2023-07-28 by lidarmansiwon

Guoc0529, 2024-09-12:
Hi, have you resolved this issue? I’m facing the same problem as well.

lidarmansiwon, 2024-09-20:
@Guoc0529 Hello! I don't remember the details but I split up the workspace and proceeded with sourcing and ran it and it worked!

Guoc0529, 2024-09-20:
> @Guoc0529 Hello! I don't remember the details but I split up the workspace and proceeded with sourcing and ran it and it worked!你好！我不记得细节了，但我把工作区分开了，然后开始采购和运行它，它奏效了！
thanks！
I don’t quite understand, and I hope you can explain a bit more in detail