koide3/hdl_graph_slam #274: map saver error

Seher-789, 2024-05-09:
when i try to save the map it is showing me error 
rosservice call /hdl_graph_slam/save_map "resolution: 0.1
> destination: '/home/seher/Project/src/mapfile.pcd'"
ERROR: Unable to communicate with service [/hdl_graph_slam/save_map], address [rosrpc://seher-Legion-Y7000P-IRH8:35245]
 this is roservice list 
rosservice list
/bag_player/get_loggers
/bag_player/set_logger_level
/gazebo/apply_body_wrench
/gazebo/apply_joint_effort
/gazebo/clear_body_wrenches
/gazebo/clear_joint_forces
/gazebo/delete_light
/gazebo/delete_model
/gazebo/get_joint_properties
/gazebo/get_light_properties
/gazebo/get_link_properties
/gazebo/get_link_state
/gazebo/get_loggers
/gazebo/get_model_properties
/gazebo/get_model_state
/gazebo/get_physics_properties
/gazebo/get_world_properties
/gazebo/pause_physics
/gazebo/reset_simulation
/gazebo/reset_world
/gazebo/set_joint_properties
/gazebo/set_light_properties
/gazebo/set_link_properties
/gazebo/set_link_state
/gazebo/set_logger_level
/gazebo/set_model_configuration
/gazebo/set_model_state
/gazebo/set_parameters
/gazebo/set_physics_properties
/gazebo/spawn_sdf_model
/gazebo/spawn_urdf_model
/gazebo/unpause_physics
/gazebo_gui/get_loggers
/gazebo_gui/set_logger_level
/hdl_graph_slam/dump
/hdl_graph_slam/load
/hdl_graph_slam/save_map
/joint_state_publisher/get_loggers
/joint_state_publisher/set_logger_level
/lidar2base_publisher/get_loggers
/lidar2base_publisher/set_logger_level
/robot_state_publisher/get_loggers
/robot_state_publisher/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rrbot/camera1/image_raw/compressed/set_parameters
/rrbot/camera1/image_raw/compressedDepth/set_parameters
/rrbot/camera1/image_raw/theora/set_parameters
/rrbot/camera1/set_camera_info
/rrbot/camera1/set_parameters
/rviz_1715259877710507026/get_loggers
/rviz_1715259877710507026/load_config
/rviz_1715259877710507026/load_config_discarding_changes
/rviz_1715259877710507026/reload_shaders
/rviz_1715259877710507026/save_config
/rviz_1715259877710507026/set_logger_level
/teleop_twist_keyboard/get_loggers
/teleop_twist_keyboard/set_logger_level
/velodyne_nodelet_manager/get_loggers
/velodyne_nodelet_manager/list
/velodyne_nodelet_manager/load_nodelet
/velodyne_nodelet_manager/set_logger_level
/velodyne_nodelet_manager/unload_nodelet

cinogul, 2024-05-14:
You need to make sure that the terminal you run hdl_graph_slam in is still running when you use rosservice call hdl_graph_slam/... In other words, open a new terminal to run the command.

Seher-789, 2024-06-03:
> You need to make sure that the terminal you run hdl_graph_slam in is still running when you use rosservice call hdl_graph_slam/... In other words, open a new terminal to run the command.

When i saved the map 
[image]
[image]

cinogul, 2024-06-04:
Without much context of what it's supposed to look like I can't help you much. It looks like there's a lot of doubled walls which means the automatic loop closure is probably failing. Try saving the bin file and using interactive SLAM to improve it.